F5OEO/rpitx #257: Python script and pocsag?

sa7bnb, 2021-05-24:
Hi, Trying to get a pocsag doorbell to my house. but I do not really succeed? 
Is there anyone who can explain why my python script does not work. 
Sorry if I post this in the wrong place! 
Regards Anders SA7BNB - Sweden

from time import sleep
import RPi.GPIO as GPIO
import time 
import os

GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(20, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)

while True:
    
    if GPIO.input(20):
        sleep(0.5)
        os.chdir("/home/pi/rpitx")
        os.system('sudo ./rpitx "1234567:Doorbell!" sudo ./pocsag -f 434.000e6')
        time.sleep(110)

VasylSamoilov, 2021-08-05:
it should be something like:
sudo printf "$ID:$MESSAGE" | ./pocsag -f $FREQUENCY -r $BAUDRATE -b $FBIT
example
printf "1234567:Doorbell!" | ./pocsag -f 434000000 -r 1200  -b 0

sa7bnb, 2021-08-05:
Solved it like this :-)
Sadly, the raspberry pi computer is not stable on this high frequency, so the temperatures cause the frequency to change slightly all the time. ;-(
Otherwise my solution worked!

This py-file to autostart on th pi

#!/usr/bin/python
import RPi.GPIO as GPIO
import os
from time import sleep

GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(20, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(24, GPIO.OUT)

while True:
    if GPIO.input(20):
        GPIO.output(24, 1)
        sleep(1)
        os.system("/home/pi/txtest.sh")
        sleep(1)
        GPIO.output(24, 0)

And this sh-file runs the command : 
#!/bin/bash
cd \rpitx
echo -e "123456:Hello World" | sudo ./pocsag -f 434.000e6

VasylSamoilov, 2021-08-05:
have you modified boot.txt? this is part of install.sh script to get pi more stable:
https://github.com/F5OEO/rpitx/blob/cce1fe6acf90d4d34ce304aed48fe80ec4ff51e7/install.sh#L38

sa7bnb, 2021-08-05:
Will check boot out, and i already using on a Raspberry pi 3 instead off pi4.