F5OEO/rpitx #259: Trying to get rpitx to compile sendiq, won't compile and leaves sendiq.cpp file in rpitx/src.

Jugga151, 2021-06-05:
I've been trying for a while now to get this compiled and installed on my pi3 b+, along with some other programs that won't compile or install, all of which use the gpio pins. Right now though, i'm just wondering how to get the install.sh to work. It properly installed all the files except for sendiq and tune. When i cd into rpitx/src and use ls -a, it shows the send iq and tune files with the sendiq and tune files with the .cpp extension and all the others are folders rather than just the one file. I'm no novice to using kali linux, but I'm also not super savvy with it yet either. I know .cpp is for c++, but I usually use python for everything I do, any help would be greatly appreciated, as I'm trying to use this to learn things like home automation and just random shit like unlocking my car with my pi3.

Jibun-no-Kage, 2022-02-18:
Just a few minutes ago, I installed RPiTX on a Model 2 B, The install script worked without issue.  All I need was the build-essential package for the compile.  Using Pi OS 11 (Bullseye 32 Bit).  Tested using sendiq to send previously reported signal for a ceiling fan light toggle, via senqiq -s 200000 -f 303.87500e6 -t u8 -i <iq file>.  Works fine.

Jibun-no-Kage, 2022-02-19:
Testing on Pi4, as well, where sendiq runs as expected, but fails to transmit correctly.  The signal is not recognized by the device it was directed to.  The same code base and iq file work fine on 2B and 3B models as noted above.  Appears the Pi4 beta still is having some issues.

Jibun-no-Kage, 2022-02-19:
Test on a different Pi 4 from the one used in the above comment.  And I get even worse results.  The first test was on a Pi 4GB RAM.  The result below is the same SD card on a Pi 4 with 8GB RAM.

`# sendiq -s 250000 -f 303.87500e6 -t u8 -i /usr/local/bin/rpitx/signals/Casitas/Light.iq
Peri Base = fe000000 SDRAM▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒<▒▒▒▒▒▒<▒▒▒<▒▒▒x▒▒▒▒▒<▒▒▒▒▒▒▒▒▒▒x<▒▒x<▒<▒▒▒▒▒▒▒▒▒▒▒▒▒x▒▒x▒▒x<x<x▒▒▒▒▒▒x<x▒▒▒▒▒▒▒<▒▒▒▒<▒▒▒▒▒▒<▒▒▒▒<▒▒xx<▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒<▒▒▒▒▒<▒▒<▒▒x▒▒▒▒x<▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒<▒▒<▒▒▒▒▒▒▒▒▒
`

I get this same garbage result, on HDMI connected monitor, or the serial console.  SSH via PuTTY does not seem to be impacted.

Via PuTTY session output...
`\# sendiq -s 250000 -f 303.87500e6 -t u8 -i /usr/local/bin/rpitx/signals/Casitas/Light.iq
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
End of file
`

But the code does not transmit in any functional manner.  So clearly some thing ugly is happening with Pi 4 beta code.  Have to reboot to recover functionality of monitor, or console session.