F5OEO/rpitx #260: Easytest Screen Blackout

HemilT, 2021-06-10:
After installing by looking at README.md, I tried cd rpitx, which worked, and the 
./easytest.sh also worked to the point where I chose the frequency, which I left default, but then when I chose FM RDS or something else on the next screen, my pi shut down or something like that. And then I had to turn it back on. Tried this multiple times with no success, please help.

HemilT, 2021-06-10:
Do I need to install a different OS, or is Raspbian Fine?

HemilT, 2021-06-10:
I am running Raspbian Buster(10).

F5OEO, 2021-06-10:
Try it without HDMI output (headless though ssh) first.

rikiberga, 2021-06-13:
@F5OEO Hi there! I am having @HemilT same issue. I can indeed confirm that using the pi headless (through ssh or RDP on windows) works fine. is just the hdmi connection that gives problems. even trying to overlock does not improve the situation. have you got any suggestions on what to do? keep it up with this GREAT project ;)

rullinoiz, 2021-07-17:
i am also having this issue, screen goes completely blank after running easysetup.sh with a raspberry pi 4. only rebooting the pi fixes the screen

TheEdgeOfFear, 2023-09-06:
Same here, no issues on zero w 2 though. I'll try on RDP. 
I did try on my zero w 2 with a RF sniffer today and there was no spike on any of the frequencies I tried.