F5OEO/rpitx #267: turns off after 5 seconds

Rafidaone, 2021-11-05:
I am using a rasberry pi 4 and wen i use easy test it will work for 5 sec and turn off my pi

micsthepick, 2021-11-05:
Do you ever see a yellow lightning bolt in the top right of the display? This sounds like a power supply issue, as the transmission might put too much load on the power supply.

radaemon, 2022-02-16:
This happens to me too as well, I have a RPI4 too.

Jibun-no-Kage, 2022-02-19:
Testing on Pi4, as well, where sendiq runs but fails to transmit correctly.  The same code base and iq file work fine on 2B and 3B models.

Rafidaone, 2022-03-04:
its not a power supply isue i have a 15 wat supply what i did find out is that it works just fine when i use console like in ssh so it seams to be a gui issue

Jibun-no-Kage, 2022-03-04:
I have been ONLY using console CLI.