F5OEO/rpitx #273: LoRa modulation/demodulation

jottr, 2022-03-06:
It would be super useful to be able to play with LoRa transmissions.
Would it be possible to implement a modulation of LoRa chirps? 

For an introduction to LoRa the [TTN website](https://www.thethingsnetwork.org/docs/lorawan/architecture/) is quite comprehensive.

creeper6530, 2024-01-09:
AFAIK the LoRa modulation and protocol are both proprietary

jottr, 2024-01-10:
Yes, it is proprietary.
I suspect that use in an open source project like this one would fall under fair-use.
I'd be happy to reach out to Semtech and ask for approval if there is interest in the community to implement this feature.

themeadery, 2025-07-28:
It's proprietary, but it has been fully reverse engineered for both TX and RX by now. Lots of resources and code scattered across the GitHub and the web to begin implementation here as well.