F5OEO/rpitx #274: DRM Digital Radio Mondiale

CowMowTV, 2022-03-15:
Hello, it would be cool if DRM could be transmitted with rpitx. Maybe with text only mode because DRM otherwise requires xHE-AAC or Ogg Opus. 

I don‘t know if the Pi could handle the required modulation/channel bandwidths but considering FreeDV runs on it, I don‘t think it will be an problem (DRM uses 4.5kHz channel bandwidth and 4/16QAM for the SDC and 16/64QAM for the MSC).

craftxbox, 2022-03-16:
OGG Opus is an open standard, It can be used in a free manner.

CowMowTV, 2022-03-17:
> OGG Opus is an open standard, It can be used in a free manner.

I didn‘t know, thanks. However to my knowledge only Dream supports DRM with Opus.

Xeddius, 2022-05-09:
you can try piping the audio output from freedv into rpitx, it should be possible using sox.