F5OEO/rpitx #275: POCSAG New Line problem

13HN1904, 2022-03-16:
I want to continue writing the text in a new line,
i.e. a line break, what do I have to do to make it work?
\n and <CR> don't work with my s.Quad X35MKV

sudo echo -e  "$ric: ABCDE \n FGAHJ " | sudo ./pocsag -f $frq -b 3 -t 4
sudo echo -e  "$ric: ABCDE <CR> FGAHJ " | sudo ./pocsag -f $frq -b 3 -t 4

VasylSamoilov, 2022-05-16:
pocsag uses 7-bit encoding, 
[image]
and pager encoding can be different from device to device.  You may try to form a message with all possible symbols in 7 bits and send it to pager, all at once or by pieces, and find out if you specific device has line break or not. Motorola pagers (pictured) don't have any, for example.

foreverlegend773633494, 2022-06-28:
Hello, friend, how can this software send Chinese to the Advisor pager?

VasylSamoilov, 2022-06-28:
POCSAG don't have a concept of code page. Just send ASCII characters and see what happens.

foreverlegend773633494, 2022-06-28:
Thank you for your reply. I tried to send ASCII characters, but Chinese characters are displayed on the pager. Whatever I send will not be what I want, SO I will first try to change the pager to English mode for testing.

VasylSamoilov, 2022-06-28:
that is correct. you can't have multiple character set at once - you need to choose will it be english or chinese. what character sets are available depends on pager rom. you need to create you own character map from chinese to ascii.