F5OEO/rpitx #278: An Update on Analog TV

03stevensmi, 2022-07-22:
Is there anyone here that can give us an update on analog pal TV support?
If not,
Any tutorials on how to test it and get it working on the raspberry pi 4?

Synthium0, 2022-08-07:
would also be interested in this

Foxy6670, 2022-09-02:
I'd like to see this work, too. It would give my old CRT a use again.
But I think you can do it using a combination of pipes, encoders, and other utilities. I have no idea how analog TV even really works.

03stevensmi, 2022-09-03:
> I'd like to see this work, too. It would give my old CRT a use again. But I think you can do it using a combination of pipes, encoders, and other utilities. I have no idea how analog TV even really works.

i may be completely bull sh*ting here but from what i know, its not the codec side thats the issue (or like you said, piping to output via encoders etc)... but the issue lies in the max frequency ( MHz) of what the raspberry pi pins can transmit.

which from what i've been reading into is nowhere near the minimum frequency that is needed to transmit pal/ntsc.
thats why its black & white with no sound, jumpy picture etc... yeah... its not 100% possible. 

HOWEVER... THIS:https://github.com/cariboulabs/cariboulite
IS VERY POSIBLE on the raspberry pi if F5OEO gets involved.
however... just be aware its not on sale and is around $119 and is 
MUCH more than the raspberry pi. if it was somthing like £25-maybe max £45-50 i would be going for this.

but without somthing around 1GHz, theres not really any better way other than  what we got now to do this sort of thing 
with only the raspberry pi itself.

but i may be wrong... ;)

Foxy6670, 2022-09-19:
> i may be completely bull sh*ting here but from what i know, its not the codec side thats the issue (or like you said, piping to output via encoders etc)... but the issue lies in the max frequency ( MHz) of what the raspberry pi pins can transmit.
> 
> which from what i've been reading into, the raspberry pi is nowhere near the minimum frequency that is needed to transmit pal/ntsc. thats why the rpitx analog tv suppirt is black & white with no sound, jumpy picture etc... yeah... its not 100% possible.

Would there be any way to shrink it down to just brightness values and sync pulses for a B&W picture? Assuming we're sticking with only 128 values of grayscale (with or without dithering but that can be quite a lot for the Pi to process) at a standard 30 or 15Hz refresh rate, that should be at most 524Hz of bandwidth, assuming each brightness value can occupy a 4Hz space (up to 512Hz) and the sync pulse takes up 12Hz at the beginning or end. Probably lackluster for the goal, but in theory it should work.

Speaking of sending data over radio, I actually did it. I sent a 12-second video with no audio, a bitrate of 4000 bit/s, a framerate of 15 and a resolution of 64x64, all by just using `minimodem` at 9600 baud and rpitx. Granted, it's VERY susceptible to corruption and data loss.

alexmyczko, 2022-10-16:
This might be of interest: https://github.com/fsphil/hacktv

03stevensmi, 2022-10-17:
> This might be of interest: https://github.com/fsphil/hacktv

I think some guy asked once if a raspberry pi would be powerfull enough to transmit using hacktv... but was (unclearly) hinted that a rpi is not powerfull enough... 

but then again, it can be used transmit on the raspberry pi with a seprerate transmitter like a soapysdr compatible device or... ofc the hackrf. 

the devs have stated they dont have a rpi so if there is some way someone could get 
the raspberry pi's rpitx (or somthing simular) transmiter supported with soapysdr
and get it to transmit on uhf and fm anything beetween 400mhz-1.2ghz...

then YES, this would work 100%.

03stevensmi, 2023-01-14:
> This might be of interest: https://github.com/fsphil/hacktv

ok, just a quick update,
hacktv has the option for raw output for use with relay attacks.
rpitx supports this.

i piped the video from hacktv into rpitx both baseband and complex on vhf 2xxmhz,
but i was notv able to pick up a picture from the pi.
however,
it did transmit.
my tv clearly picked up what it was transmitting but i think due to the pixelrate and samplerate
from my rpi2b, it stated it was to low.
i did moderfy rpitx to bypass this aand that worked, but just heavy distortion.

i tried float, int etc etc but i got the signal, but no color or picture.

i used a rpi2b and havn't got around to using my pi4, but if anyone knows if this can be possible,
please feel free to help us all out and give it a go to see what comes out.

cheers

Foxy6670, 2023-02-12:
@03stevensmi so you got it to transmit using the full 6 MHz of bandwidth required? How did you go about doing this? I'd love to know - I would like to give my old TV a use.

03stevensmi, 2023-02-12:
> @03stevensmi so you got it to transmit using the full 6 MHz of bandwidth required? How did you go about doing this? I'd love to know - I would like to give my old TV a use.

i never said i got a picture, just a signal.
Funny thing is i accually spoke to phil (hacktv) on discord and well... he sadly stated that the rpi4 hasn't got enough bandwidth for most tv signals. So its looking like the hacktv is a no go on the rpi. however... i DID get a picture with rpitx analog tv sooo..... it may be possible with b&w. it was a bit scrambled but it was there. Unfortunatly rptx only coded it with a B&w colour bar and no video pass through so anything regarding transmitting analog tv on the rpi has only been done with rpitx. I still feel that there is a way around this... but how anybody gets around in solving this feature... idk if anybody now days is clever enough to know how. F5OEO and other smart coders like him are fading away fast, as is rpi's from what i heard. so if anyone has any smarts in this kind of stuff... feel free to fork this and give it a go. But as for now, it looks like F5OEO is (was) the only person that has the skill in making this possible.

byronbonkers, 2023-07-30:
could u add me on discord, cus I would love to know how u did it? my username is byronbonkers