F5OEO/rpitx #285: about rpi4 crash when rpitx run

sunnychon, 2022-09-25:
you just need to open terminal

type sudo raspi-config

select "2 Display Options"

select "D1 Resolution"

select "mode 14: 720x480 @ 60Hz 4:3, clock:5mode 15: 720x480 @ 60Hz 16:9, clock:"

and reboot

logianer, 2023-07-30:
I got a rpi4b and I don't get the option D1 Resolution. I have raspi-config on the latest version. Am I doing something wrong?

logianer, 2023-07-30:
> select "D1 Resolution"
> 
> select "CEA Mode 4"

For everone that doesn't have these options, add these lines at the bottom of `/boot/config.txt`:
```yml
hdmi_group=1
hdmi_mode=4
```

TheEdgeOfFear, 2023-09-05:
I am having the exact same issue:
added the above lines to the config file, 
hdmi_group=1
hdmi_mode=4

sudo apt update
sudo apt upgrade
sudo rpi-update,

GPU Temp --- temp=53.5'C

Error Log:
Sep 06 09:14:05 raspberrypi dbus-daemon[472]: [system] Successfully activated s>
Sep 06 09:14:05 raspberrypi systemd[1]: Started Daemon for power management.
░░ Subject: A start job for unit upower.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit upower.service has finished successfully.
░░ 
░░ The job identifier is 1224.
Sep 06 09:14:22 raspberrypi kernel: [UFW BLOCK] IN=eth0 OUT= MAC=01:00:5e:00:00>
Sep 06 09:14:22 raspberrypi kernel: [UFW BLOCK] IN=wlan0 OUT= MAC=01:00:5e:00:0>
Sep 06 09:14:26 raspberrypi kernel: [UFW BLOCK] IN=eth0 OUT= MAC=01:00:5e:00:00>
Sep 06 09:14:26 raspberrypi kernel: [UFW BLOCK] IN=wlan0 OUT= MAC=01:00:5e:00:0>
Sep 06 09:16:09 raspberrypi kernel: v3d fec00000.v3d: MMU error from client L2T>
Sep 06 09:16:27 raspberrypi kernel: [UFW BLOCK] IN=eth0 OUT= MAC=01:00:5e:00:00>
Sep 06 09:16:27 raspberrypi kernel: [UFW BLOCK] IN=wlan0 OUT= MAC=01:00:5e:00:0>
Sep 06 09:16:32 raspberrypi kernel: [UFW BLOCK] IN=eth0 OUT= MAC=01:00:5e:00:00>
Sep 06 09:16:32 raspberrypi kernel: [UFW BLOCK] IN=wlan0 OUT= MAC=01:00:5e:00:0>
Sep 06 09:17:01 raspberrypi CRON[1932]: pam_unix(cron:session): session opened >
Sep 06 09:17:01 raspberrypi CRON[1933]: (root) CMD (   cd / && run-parts --repo>
Sep 06 09:17:01 raspberrypi CRON[1932]: pam_unix(cron:session): session closed >
lines 3735-3757/3757 (END)

I can get the program running on a Zero W without crashing but ideally I would like it running on a 4B.

Any help or suggestion cheers....