F5OEO/rpitx #289: Not an issue: I've started experiments and wish to establish contacts

giorgio-denunzio, 2022-12-11:
Hello, I wish to understand if rpitx (great idea and software!) can be actually used for HF TX in HAM bands, SSB and CW.
I collected as much info as possible and I have just started my tests (essentially a check of spurs and harmonics). As everyone, I am struggling with filtering (not built/bought anything yet: just studying). Amplification will follow.
If interested in collaborating, or if you have working solutions, or if you know that any solution would be sub-optimal and de facto useless for practical usage, please contact me, here or possibly at my email address  giorgio.denunzio (at) unisalento.it.
Thanks
Giorgio IU7RAW

sizzlik, 2023-11-02:
You want to use a rpi with the tx power lower then a car keyfob to transmit to Ham Radios? Just buy a SDR that can transmit and some cheap rf amps and most important a good antenna. What you ask is like playing real golf with minigolf equiptment. Rpitx is more of a proof of concept/prank software then something you can do serious RF stuff with. Get a hackrf clone from aliexpress, amps, bandpass filter and an antenna, get up on a hill on a clear day and you might tx a few kilometers. Plus, if you dont have a radio licencens..stay the F out. You can listen, but dont try to transmit on the ham bands if you dont even know what you do...or do you want to risk disturbing air traffic? Read, learn, read some more and then make a license for ham radios.

giorgio-denunzio, 2023-11-02:
Dear Moe,
thanks for your message.
Let me briefly reply.
Before all, as I made some further experiment, at present I tend to 
agree that "Rpitx is more of a proof of concept", because reliability is 
not high: during my tests I was having a good carrier and reasonable 
modulation, then I installed something on the rpi, and rpitx stopped 
working at all (more exactly, it worked, no error, but no carrier was 
emitted). Of course the responsible was not rpitx (which is an 
extraordinary piece of software) but my raspberry pi and its operating 
system. Anyway, after this issue I temporarily gave up (and began to 
experiment with adalm pluto), but maybe one day I will restart... It is 
so intriguing!
That said, I will be a bit critical towards your message, please excuse 
me. You say that rpi tx power is very low and this is well known: of 
course one has to filter, amplify, maybe filter again before going on air.
You also cite hackrf, but then you (of course) say you have to amplify 
its output too, so the principle is the same (the big difference being 
that hackrf was created to emit rf while raspberry pi was not). But this 
is what I like: experimenting! I am currently building a lpf to be used 
with pluto and/or my rpi.
Where I really see you going off on a tangent (please, really excuse me 
for being a bit rude) is where you start with "Plus, if you dont have a 
radio licencens..stay the F out. You can listen, but dont try to 
transmit on the ham bands if you dont even know what you do...or do you 
want to risk disturbing air traffic? Read, learn, read some more and 
then make a license for ham radios."
May I feel a little bit offended? What makes you think I do not know 
what I do? Please reread what you wrote, word by word, "F" included, 
reflect a couple of minutes, then go back to my original message and 
check my signature: Giorgio IU7RAW. It was even in capital letters... 
Anyway, I agree, I will read and learn and read again... it is one of 
life pleasures!
Best regards
Giorgio IU7RAW

-- 
Giorgio De Nunzio PhD
Researcher and adjunct professor
Dipart. di Matematica e Fisica "Ennio De Giorgi", Univ. del Salento
INFN (Istituto Nazionale di Fisica Nucleare)
DReAM (Laboratorio Diffuso di Ricerca Interdisciplinare Applicata alla Medicina)
Lecce - Italy
***@***.***
tel +39 0832 297084-297051
mobile +39 320 3829845
http://orcid.org/0000-0002-1998-0286
Scopus ID: 6603335538
WOS Researcher ID: L-7283-2015
Google Scholar ID: fOyVq5sAAAAJ
Amateur radio call sign: IU7RAW
"The whole problem with the world is that fools and fanatics are always so certain of themselves, but wiser people so full of doubts."
Bertrand Russell