F5OEO/rpitx #291: Alternative SBC while Raspberry Pi's are hard to obtain?

Trevo525, 2023-01-04:
It's pretty difficult to get a Raspberry Pi currently. There are several alternatives like Rock Pi, Orange Pi, or Odroid. I saw that on [this issue](https://github.com/F5OEO/rpitx/issues/39) that this repo is pretty specific to Raspberry Pi. Is there any alternatives? I have something I want to transmit that isn't using 433 (304.3 is what it is using). 

I have been able to use a Raspberry Pi that I own to test this repo and it was successful. But, I need that Pi for other things and I need a solution to transmit. If I could use an esp8266 or esp32, that would be ideal.

CurtisDufour, 2023-02-17:
I would be interested in this as well. I've puzzled over how I could get my Orange Pi 4 LTS, or even my Le Potato Libre Computer to utilize this repo, but I wouldn't know where to start with making that happen on my own.

F5OEO, 2024-02-26:
Again, this project use very special functionalities of raspberry SOC. Other clone could not be used

closed 2024-02-26 by F5OEO