F5OEO/rpitx #293: SendIQ or RTL-SDR command not working properly

Mifaxav, 2023-01-19:
recorded iq file using rtl sdr in 2 different frequencies. Sending it using sendiq makes me get the same output of a 0.3 second bump in the radio spectrum and gone. One of the frequencies had a car key fob recorded with the other being megahertz away. I have a Raspberry pi 3B+ and i truly dont know whats going on

darksidelemm, 2023-06-21:
Interestingly I get similar results with sendiq and the transponder code. Platform is a RPI 3B+
Short carrier transmission, then nothing. 

I've tried quite a few variations, all with pretty similar results. I get transmission of a carrier for a while then nothing, or sometimes I just get nothing at all. 

I've also tried with a few other variations, to try and get somewhat more narrowband output:

* `rtl_fm -M raw -f 146500000 -s 96000 -F9 - | sudo ./sendiq -s 96000 -f 444000000 -t i16 -i /dev/stdin` - Short carrier on 444 MHz, then nothing.
* `rtl_fm -M raw -f 146500000 -s 96000 -F9 - | buffer | sudo ./sendiq -s 96000 -f 444000000 -t i16 -i /dev/stdin` - Same result as above.
* `cat /dev/urandom | sudo ./sendiq -s 96000 -f 444000000 -t i16 -i /dev/stdin` - This produces RF output at 444 MHz continuously.

This makes me think the issue is something to do with stdin input into sendiq, but there's not really anything obvious in the output that explains if this is the case or not. 

As a test I also tried:
```
rtl_fm -M raw -f 146500000 -s 96000 -F9 - > test.bin
```
(and waited a while, resulting in a ~16Mb file)

... then
```
cat test.bin | sudo ./sendiq -s 96000 -f 444000000 -t i16 -i /dev/stdin
```
... this resulting in a very short 'blip' at 444 MHz, then nothing. 

Running:
```
sudo ./sendiq -s 96000 -f 444000000 -t i16 -i test.bin
```
Results in the same effect, so maybe this rules out stdin issues, and instead there is just some issue with librpitx and/or sendiq on the platform I'm running?

As extra info:
* Raspbian Bullseye
* `uname -a` = Linux cameratest 6.1.25-v7+ #1647 SMP Wed Apr 26 17:35:40 BST 2023 armv7l GNU/Linux

darksidelemm, 2023-06-22:
Similar behaviour on Raspbian Buster: Linux rpitx 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

Mifaxav, 2023-06-24:
i got the issue fixed by giving a higher samplerate. Try 960000 (not really a pro at this stuff lol)

darksidelemm, 2023-06-24:
A higher samplerate doesnt really solve the issue for me! Also, given the example transponder code isn't working, that's probably something that should be looked into..

Mifaxav, 2023-06-24:
then i have no idea what to do :( 

Good luck anyway... 

maybe try looking at how other people have done it and like get a better idea?

NoahFix, 2026-10-09:
I was looking through the `librpitx` source code.**I suspect that rpitx and librpitx may share some of the same underlying logic**.And I  might have a potential explanation/hypothesis for the issue you encountered.

Potential Cause:

When `sendiq` reads IQ samples, it converts them into polar coordinates to control hardware:

(https://github.com/F5OEO/librpitx/blob/master/src/dsp.cpp line75)
```cpp
amplitude = abs(sample); 
int IntAmplitude = (int)(mydsp.amplitude * 8.0) - 1;
```

Because the Pi's GPIO drive-strength control only has 8 discrete levels (3-bit), `librpitx` seems to have mapped the normalized amplitude `[0.0, 1.0]` to values from `0` to `7`.

There is a check in the codebase(https://github.com/F5OEO/librpitx/blob/master/src/iqdmasync.cpp line140 to 157):

```cpp
int IntAmplitude = (int) (mydsp.amplitude * 8.0) - 1;
// ...

int IntAmplitudePAD = IntAmplitude;
if (IntAmplitude > 7) {
    IntAmplitudePAD = 7;
}
if (IntAmplitude < 0) {
    IntAmplitudePAD = 0;
    IntAmplitude = -1; // leads to mute
}
// ...

if(IntAmplitude==-1) {
  sampletab[Index*registerbysample+2]=(Originfsel & ~(7 << 12)) | (0 << 12); //Pin is in -> Amplitude 0
} else {
  sampletab[Index*registerbysample+2]=(Originfsel & ~(7 << 12)) | (4 << 12); //Alternate is CLK : Fixme : do not work with clk2
}
```

When `IntAmplitude` becomes `-1` , `librpitx` dynamically switches the GPIO pin back to Input mode via DMA. This disconnects the RF output pin from GPCLK to achieve a true mute.

If the incoming IQ stream from `rtl_fm` has relatively low gain, `sendiq` will choose to ignore the weak signal possibly because the GPIO of rpi can't output such a low voltage.So the signal disappeared suddenly

Possible Solution:

1. Increase RTL-SDR Gain:

   If you're using `rtl_fm`, you could try adding an explicit high gain flag (e.g., `-g 24` ).Don't use too big gain because it may lead to overload of your RTL-SDR

   ```
   rtl_fm -M raw -g 24 -f 146500000 -s 96000 -F9
   ```

2. Normalize Recorded IQ Files:

   For pre-recorded `.bin` files, scaling/normalizing the I/Q samples so their peak values utilize more of the `i16` range ($\pm 32767$) might help prevent it from dropping below the threshold.

Hope this might be helpful!