F5OEO/rpitx #294: How To Send Data and Image using rpitx?

nau4k, 2023-02-11:
Hello everyone, I have a project from my lecturer. this project uses Raspberry pi 3b+ and RTL-SDR.
Raspberry using rpitx to transmit RF signal.
The expected result is that raspberry can send data and images via Radio Frequency and received with rtl sdr with gnu radio. Hopefully anyone can teach me i will be very grateful

radiocolin, 2023-02-12:
So you want someone else to do your project?

nau4k, 2023-02-14:
> So you want someone else to do your project?

I'm not sure about that, I just want to know what to use to do the project

Lucstay11, 2023-02-24:
> Hello everyone, I have a project from my lecturer. this project uses Raspberry pi 3b+ and RTL-SDR. Raspberry using rpitx to transmit RF signal.
> The expected result is that raspberry can send data [...]

A can do it for you!

nau4k, 2023-02-24:
> A can do it for you!

can you teachme please?

Lucstay11, 2023-02-24:
Use the 5th option of rptix you can send images via SSTV.
_If that's just what you want to do, it's not very complicated_

mattvall1, 2023-02-28:
> Hello everyone, I have a project from my lecturer. this project uses Raspberry pi 3b+ and RTL-SDR. Raspberry using rpitx to transmit RF signal. The expected result is that raspberry can send data [...]

Completely out of interest, what degree are you studying for. That sounds very cool!

closed 2023-05-29 by nau4k