F5OEO/rpitx #295: Install for Alpine Linux

AlexCzar, 2023-02-12:
I'm running Alpine on my rpi 2B, can rpitx be installed there, search came up empty :|

mfalkvidd, 2023-02-13:
What happened when you tried?

AlexCzar, 2023-02-13:
I haven't, it wouldn't have worked if I did, the first thing the installation script does is installing packages with apt, which isn't present on Alpine, I could follow the script manually, but there are some packages needed which I'm unfamiliar with and which aren't present in Alpine, at least by the name in the installation script.

mfalkvidd, 2023-02-13:
Could you list the missing packages? Maybe someone knows more about them.

AlexCzar, 2023-02-13:
While writing an answer to your question, I did some more investigating and found the corresponding packages.
Now I'm having issue compiling:

```bash
; git clone https://github.com/F5OEO/csdr
; make
```
<details><summary>Output</summary>

```
g++ -O3 -ffast-math -fdump-tree-vect-details -dumpbase dumpvect -mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4 -funsafe-math-optimizations -Wformat=0 nmux.cpp tsmpool.cpp -g -lm -lrt -lfftw3f -DUSE_FFTW -DLIBCSDR_GPL -DUSE_IMA_ADPCM -L. -lcsdr -lpthread -Wno-unused-result -o nmux
nmux.cpp: In function 'int main(int, char**)':
nmux.cpp:155:9: error: 'fd_set' was not declared in this scope
  155 |         fd_set select_fds;
      |         ^~~~~~
nmux.cpp:179:26: error: 'select_fds' was not declared in this scope
  179 |                 FD_ZERO(&select_fds);
      |                          ^~~~~~~~~~
nmux.cpp:179:17: error: 'FD_ZERO' was not declared in this scope
  179 |                 FD_ZERO(&select_fds);
      |                 ^~~~~~~
nmux.cpp:180:17: error: 'FD_SET' was not declared in this scope; did you mean 'L_SET'?
  180 |                 FD_SET(listen_socket, &select_fds);
      |                 ^~~~~~
      |                 L_SET
nmux.cpp:185:34: error: 'select' was not declared in this scope
  185 |                 int select_ret = select(highfd, &select_fds, NULL, NULL, NULL);
      |                                  ^~~~~~
nmux.cpp:190:21: error: 'FD_ISSET' was not declared in this scope; did you mean 'CPU_ISSET'?
  190 |                 if( FD_ISSET(listen_socket, &select_fds) && ((new_socket = accept(listen_socket, (struct sockaddr*)&addr_cli, &addr_cli_len)) != -1) )
      |                     ^~~~~~~~
      |                     CPU_ISSET
nmux.cpp:246:21: error: 'FD_ISSET' was not declared in this scope; did you mean 'CPU_ISSET'?
  246 |                 if( FD_ISSET(STDIN_FILENO, &select_fds) )
      |                     ^~~~~~~~
      |                     CPU_ISSET
make: *** [Makefile:60: nmux] Error 1
```
</details>

AlexCzar, 2023-02-13:
Looks like the problems are either caused by incompatibilities with musl, or maybe missing dependency which isn't mentioned in the install.sh script, I don't know enough about C/C++/libc ecosystem to figure it out further.