F5OEO/rpitx #308: Do we support the Pico?

fumin, 2023-10-23:
In the README, we list a few rpi devices, but the Pico nor the Pico W is not there.
Do we support them?

If we are, on the Pico, should we be using pin GP4 (pin 6 in the below picture)?

https://www.raspberrypi.com/documentation/microcontrollers/images/picow-pinout.svg

fumin, 2023-10-23:
@kj7rrv thanks for your explanations.
In this case, I'm probably going to get a Pi Zero W.

Here's a related question, the README mentions that the Zero W is supported.
I wonder how about the Zero 2 W (the 2nd generation)? Do we support the Zero 2 W?

F5OEO, 2024-02-26:
Even rasberry pico and raspberry zero are closed in name, this is a complete different architecture. Some rpitx equivallent on pico exists with lower frequencies, but the architecture is completely different. So not relevant here !

closed 2024-02-26 by F5OEO