F5OEO/rpitx #312: Pager message sent 4 times on some frequencies

mrDevGit, 2024-01-01:
Hi 

Love the RPITX. However, on some frequencies the message is sent  (or received?) 4 times. Some pagers show 4 seperate messages others just show one and flag as a duplicate. 

When I send the same pager a message via the HackRF One, it is sent once, as expected. 

My setup:
Pi 4
15cm Dupont wire on GPIO4

Pagers affected:
All above 400mhz as far as I can tell. lower frequencies receive the message once as expected.

Any thoughts?

Thanks

Example output:

echo -e "1234567:Test 123" | sudo ./pocsag -f "929162500" -b 3
DEBUG DATA = I=0   P=0x1e730a8 T=0 L=35
DEBUG DATA = I=1   P=0x1e74150 T=35 L=17
DEBUG DATA = I=2   P=0x1e74228 T=52 L=17
DEBUG DATA = I=3   P=0x1e74348 T=69 L=17
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
dma PI4 using channel 7
channel 7 CBSize 24006 UsermemSize 12003
Using mbox device /dev/vcio.
816204 Size NUM PAGES 200 PAGE_SIZE 4096
BufferSize 12003 , cb 24006 user 12003
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Info:NTP find offset -121309 freq -909944 pps=0 ppm=-13.884644
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Info:NTP find offset -121309 freq -909944 pps=0 ppm=-13.884644
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
Info:NTP find offset -121309 freq -909944 pps=0 ppm=-13.884644
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
PLLC_DIG0=006bd422
PLLC_DIG1=0002401a
PLLC_DIG2=00400000
PLLC_DIG3=00000002
PLLC_ANA0=00000000
PLLC_ANA1=0011c000
PLLC_ANA2=00d00000
PLLC_ANA3=00000052
PLLC_DIG0R=006bd422
PLLC_DIG1R=0002401a
PLLC_DIG2R=00400000
PLLC_DIG3R=00000002
PLLA_ANA0=00000000
PLLA_ANA1=0011c000 prediv=1
PLLA_ANA2=00d04000
PLLA_ANA3=00000052
GNRIC CTL=0000636d DIV=    636d  VPU   CTL=00000214 DIV=    4000
SYS   CTL=00000040 DIV=       0  PERIA CTL=00000040 DIV=       0
PERII CTL=00000040 DIV=       0  H264  CTL=00000244 DIV=    1000
ISP   CTL=00000244 DIV=    1000  V3D   CTL=000002d4 DIV=    1000
CAM0  CTL=00000200 DIV=       0  CAM1  CTL=00000200 DIV=       0
CCP2  CTL=0000636d DIV=    636d  DSI0E CTL=00000200 DIV=       0
DSI0P CTL=00000000 DIV=       0  DPI   CTL=00000200 DIV=       0
GP0   CTL=00000005 DIV=    2000  GP1   CTL=00000200 DIV=       0
GP2   CTL=00000200 DIV=       0  HSM   CTL=0000636d DIV=    636d
OTP   CTL=00000091 DIV=    4000  PCM   CTL=00000296 DIV=  ff504a
PWM   CTL=00000200 DIV=       0  SLIM  CTL=0000636d DIV=    636d
SMI   CTL=00000200 DIV=       0  SMPS  CTL=0000636d DIV=    636d
TCNT  CTL=00000000 DIV=     9c4  TEC   CTL=00000000 DIV=       0
TD0   CTL=00000200 DIV=       0  TD1   CTL=00000200 DIV=       0
TSENS CTL=00000091 DIV=   10000  TIMER CTL=00000291 DIV=   36000
UART  CTL=00000296 DIV=    fa00  VEC   CTL=00000000 DIV=       0
PULSE CTL=00000091 DIV=   1b000  PLLT  CTL=00000000 DIV=????????
DSI1E CTL=00000200 DIV=       0  DSI1P CTL=00000098 DIV=       0
AVE0  CTL=00000000 DIV=       0
CMPLLA=00000000  CMPLLC=0000022a 
CMPLLD=00000000   CMPLLH=0000636d 
EMMC  CTL=00000296 DIV=    3000
EMMC  CTL=00000296 DIV=    3000
EMMC  CTL=00000296 DIV=    3000
PLLA PDIV=1 NDIV=60 FRAC=194181   3249.999962 MHz
DSI0=0 CORE=13 PER=256 CCP2=0

PLLB PDIV=1 NDIV=55 FRAC=582542   3000.000092 MHz
ARM=2 SP0=6369911 SP1=256 SP2=256

PLLC PDIV=1 NDIV=34 FRAC=432885   1858.292839 MHz
CORE2=0 CORE1=0 PER=2 CORE0=256

PLLD 21037 PDIV=1 NDIV=55 FRAC=582544   3000.000092 MHz
DSI0=0 CORE=5 PER=4 DSI1=0

PLLH PDIV=0 NDIV=256 FRAC=256   13824.013184 MHz
AUX=0 RCAL=0 PIX=6369911 STS=6369911

Info:NTP find offset -121309 freq -909944 pps=0 ppm=-13.884644
Pi4=1 Xosc = 54000000 Freq PLL no 5= 929159345
Found PLL solution for frequency 1858.3MHz : divider:1 VCO: 3716.6MHz
Pll Fix Divider
Use PLL Prediv
Master PLLC Locked
Info:NTP find offset -121309 freq -909944 pps=0 ppm=-13.884644
Pi4=1 Xosc = 54000000 Freq PLL no 6= 750010435
PCM prediv = 153
PCM clk=4085 / 98
FSK Origin fsel 0
FSK burst end Tx

mfalkvidd, 2024-01-01:
https://github.com/F5OEO/rpitx/blob/master/src/pocsag/pocsag.cpp#L574 sets `int REPEAT_COUNT = 4;`

I guess you'll want to add `-t1` to the command line to only send once?

Also, since you seem to have missed this part of the [readme](https://github.com/F5OEO/rpitx/blob/master/README.md#hardware):
[image]

mrDevGit, 2024-01-01:
Thanks! I am very very new to this.... how do i create a band pass filter? Could you point me at a link? Stupid question in the spirit of learning: Is that a software of hardware filter?

mfalkvidd, 2024-01-01:
Thanks for asking.
It is a hardware filter. They can be bought pre-made, example https://www.ebay.com/itm/164586150463 or built do-it-yourself: https://alicja.space/blog/designing-lna-with-bandpass-filter/

Without it, rpitx will send on many other frequencies (other than the intended frequency), often including military and cellular frequencies which can cause lots of problems.