F5OEO/rpitx #318: Transmission not working on Raspberry Pi 3B+

marines1, 2024-02-05:
I installed rpitx on Raspbian, but transsmision not working. I tried easytest.sh with difference options  with various options (carrier,chirp etc.)     but no transmission. I listens to it at different frequencies (RtlSDR on PC) but there is silence. I tested PiFmRds earlier and everything worked there  , but PiFmRds have frequency limit.

IrfanNafiz, 2024-02-07:
Same issue, I'm testing on a Rpi3B

F5OEO, 2024-02-26:
Which OS are you using ? Model of pi is not enough to troubleshoot.

B-Junk, 2024-02-27:
Same issue, although I WAS able to get a carrier using the "chirp" command at 105.9 and 462 MHz.
Fresh install using Raspberry Pi Imager (no desktop) on a PI 3 B.  Followed the instructions on the:
https://github.com/F5OEO/rpitx/blob/master/README.md

cat /etc/os-release:
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```

B-Junk, 2024-02-27:
Install.sh had the following errors:
`sendiq.cpp: In function ‘int main(int, char**)’:
sendiq.cpp:132:70: error: expected ‘;’ before ‘exit’
  132 |         fprintf(stderr, "tune: not a valid frequency - '%s'", optarg)
      |                                                                      ^
      |                                                                      ;
  133 |         exit(1);
      |         ~~~~
make: *** [Makefile:40: ../sendiq] Error 1
g++ -std=c++11 -Wall -g -O2 -Wno-unused-variable -o ../sendiq sendiq.cpp  -L/opt/vc/lib -lrpitx -lm -lrt -lpthread
sendiq.cpp: In function ‘int main(int, char**)’:
sendiq.cpp:132:70: error: expected ‘;’ before ‘exit’
  132 |         fprintf(stderr, "tune: not a valid frequency - '%s'", optarg)
      |                                                                      ^
      |                                                                      ;
  133 |         exit(1);
      |         ~~~~
make: *** [Makefile:40: ../sendiq] Error 1

In order to run properly, rpitx need to modify /boot/config.txt. Are you sure (y/n)
`

B-Junk, 2024-02-28:
After playing "wack-a-mole" I found similar missing semicolon errors in tune.cpp and rpitx.cpp.  I now can run testssb.sh for example, but some others in the test script do not work.

IrfanNafiz, 2024-02-28:
Apologies for the vagueness. I was in a rush to get a Tx project working. 

I used rpi3b and flashed a Bookworm 32bit OS using RpiImager. Even after all the requirements were installed and rebooting, rpitx was not transmitting. I tested 2-3 times to confirm it wasn't working using an RTL-SDR to check for the Tx.

But here's the catch.

I shifted to fm_transmitter project (which worked) and installed dependencies for that - did some experimenting with modules related to lora, and just happened to try out rpitx again - and this time rpitx worked. So I'm not sure what went wrong initially. 

The gaps between my initial and final trial was 2-3 days so it's hard for me to figure out what i did in between. If anyone can recreate this maybe it'll be helpful.

F5OEO, 2024-03-11:
Should be fixed with latest commits. Reopen if still issue

closed 2024-03-11 by F5OEO