F5OEO/rpitx #323: Raspbery Pi Bookworm lbcm_hostand semicolons missing

ChazTuna, 2024-03-06:
'librpitx' compile issue:
g++ -fPIC -shared -o librpitx.so mailbox.o raspberry_pi_revision.o rpi.o amdmasync.o atv.o dma.o dsp.o fmdmasync.o fskburst.o gpio.o iqdmasync.o ngfmdmasync.o ookburst.o phasedmasync.o serialdmasync.o util.o -lm -lrt -lpthread -L/opt/vc/lib -lbcm_host -fPIC
/usr/bin/ld: cannot find -lbcm_host: No such file or directory

Not there on the standard 32 bit install.
A package was missing that is required. it is contained in libraspberrypi-dev

Fixed by “sudo apt install libraspberrypi-dev” ahead of make.
*******************
Semicolon missing in sendiq.cpp, rpitx.cpp, tune.cpp after fprint statements: (see Pull request on this)
 for instance:    fprintf(stderr, "tune: not a valid frequency - '%s'", optarg)

******************
Also:
g++ -std=c++11 -Wall -g -O2 -Wno-unused-variable -o ../pirtty pirtty/pirtty.cpp  -L/opt/vc/lib -lrpitx -lm -lrt -lpthread
install -m 0755 ../pisstv
install: missing destination file operand after '../pisstv'
******************
Also lots of "warnings"
corel8/corel8.cpp:33:57: warning: array subscript has type ‘char’ [-Wchar-subscripts]
   33 |             fprintf(stderr,"freq %d -> %c\n",4*(Costas8[Symbol][i]-1),Symbol);
corel8/corel8.cpp:35:44: warning: array subscript has type ‘char’ [-Wchar-subscripts]
   35 |                 Tab[i*Upsample+j]=(Costas8[Symbol][i]-1);
morse/morse.cpp: In function ‘void morse_to_cw(const char*)’:
morse/morse.cpp:108:14: warning: variable ‘cw’ set but not used [-Wunused-but-set-variable]
  108 |         char cw[23];// = "mmmmmmmmmmmmmmmmmmmmmmm";
morse/morse.cpp: In function ‘int main(int, char**)’:
morse/morse.cpp:228:27: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘unsigned int’} [-Wsign-compare]
  228 |         for (int i = 0; i < strlen(msg); i++)

coolest42u, 2024-03-08:
g++ -std=c++11 -Wall -g -O2 -Wno-unused-variable -o ../pirtty pirtty/pirtty.cpp -L/opt/vc/lib -lrpitx -lm -lrt -lpthread
install -m 0755 ../pisstv
install: missing destination file operand after '../pisstv'

+1

F5OEO, 2024-03-11:
Fixed with https://github.com/F5OEO/rpitx/commit/f0da813fe7875d21ac81bef34ee2f8e7c3850586

closed 2024-03-11 by F5OEO