F5OEO/rpitx #325: dvb/dvbsenco8.s:2: Error: junk at end of line, first unrecognized character is `@'

epif4nio, 2024-03-29:
Hi, I'm trying to build the project using a **Raspberry Pi 5** with **Raspberry Pi OS (64-bit)** but I'm getting some errors.

There are a lot of errors that look like this:
```
dvb/dvbsenco8.s:2: Error: junk at end of line, first unrecognized character is `@'
```

Here is the full installation log:

```
Install rpitx - some package need internet connection -
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
libsndfile1-dev is already the newest version (1.2.0-1).
git is already the newest version (1:2.39.2-1.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
imagemagick is already the newest version (8:6.9.11.60+dfsg-1.6+deb12u1).
libfftw3-dev is already the newest version (3.3.10-1).
libraspberrypi-dev is already the newest version (1:2+git20231018~131943+3c97f76-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
rtl-sdr is already the newest version (0.6.0-4).
buffer is already the newest version (1.19-12+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'csdr'...
NOTE: you may have to manually edit Makefile to optimize for your CPU (especially if you compile on ARM, please edit PARAMS_NEON).
Auto-detected optimization parameters: -mcpu=cortex-a53 -mtune=cortex-a53 -funsafe-math-optimizations -Wformat=0

rm -f dumpvect*.vect
gcc -std=gnu99 -O3 -ffast-math -fdump-tree-vect-details -dumpbase dumpvect -mcpu=cortex-a53 -mtune=cortex-a53 -funsafe-math-optimizations -Wformat=0 fft_fftw.c libcsdr_wrapper.c  -g -lm -lrt -lfftw3f -DUSE_FFTW -DLIBCSDR_GPL -DUSE_IMA_ADPCM -Wno-unused-result -fpic -shared -Wl,-soname,libcsdr.so.0.15 -o libcsdr.so.0.15
./parsevect dumpvect*.vect
/bin/sh: 1: ./parsevect: not found
make: [Makefile:52: libcsdr.so] Error 127 (ignored)
gcc -std=gnu99 -O3 -ffast-math -fdump-tree-vect-details -dumpbase dumpvect -mcpu=cortex-a53 -mtune=cortex-a53 -funsafe-math-optimizations -Wformat=0 csdr.c -g -lm -lrt -lfftw3f -DUSE_FFTW -DLIBCSDR_GPL -DUSE_IMA_ADPCM -L. -lcsdr -Wno-unused-result -o csdr
In file included from /usr/include/aarch64-linux-gnu/bits/libc-header-start.h:33,
                 from /usr/include/stdio.h:27,
                 from csdr.c:34:
/usr/include/features.h:194:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
  194 | # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
      |   ^~~~~~~
csdr.c:163:6: warning: trigraph ??< ignored, use -trigraphs to enable [-Wtrigraphs]
  163 | "    ??<jump_to_function_docs_on_github>\n"
      |       
g++ -O3 -ffast-math -fdump-tree-vect-details -dumpbase dumpvect -mcpu=cortex-a53 -mtune=cortex-a53 -funsafe-math-optimizations -Wformat=0 nmux.cpp tsmpool.cpp -g -lm -lrt -lfftw3f -DUSE_FFTW -DLIBCSDR_GPL -DUSE_IMA_ADPCM -L. -lcsdr -lpthread -Wno-unused-result -o nmux
install -m 0755 libcsdr.so.0.15 /usr/lib
install -m 0755 csdr /usr/bin
install -m 0755 csdr-fm /usr/bin
install -m 0755 nmux /usr/bin
#-install -m 0755 ddcd /usr/bin
Cloning into 'librpitx'...
g++ -Wall -O3 -Wno-unused-variable -std=c++11 -fPIC -c mailbox.c
g++ -Wall -O3 -Wno-unused-variable -std=c++11 -fPIC -c raspberry_pi_revision.c
raspberry_pi_revision.c: In function ‘int getRaspberryPiInformationForRevision(int, RASPBERRY_PI_INFO_T*)’:
raspberry_pi_revision.c:459:33: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
  459 |                 if (memoryIndex < knownMemoryValues)
      |                     ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
raspberry_pi_revision.c:471:36: warning: comparison of integer expressions of different signedness: ‘int’ and ‘size_t’ {aka ‘long unsigned int’} [-Wsign-compare]
  471 |                 if (processorIndex < knownProcessorValues)
      |                     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
[... 642 lines cut ...]
dvb/dvbsenco8.s:346: Error: operand 1 must be an integer register -- `mov r6,#IPACKETSIZE'
dvb/dvbsenco8.s:347: Error: operand 1 must be a SIMD vector register -- `mul r5,r6@:r5=offset into ipackets of first byte to be filled'
dvb/dvbsenco8.s:348: Error: operand 1 must be an integer register -- `mov r6,#204'
dvb/dvbsenco8.s:350: Error: operand 1 must be an integer register -- `ldrh r7,[r3],#2@:get a 16 bit offset value from the interleave table and increment'
dvb/dvbsenco8.s:351: Error: operand 1 must be an integer register -- `ldrb r8,[r0],#1@:get an input byte'
dvb/dvbsenco8.s:352: Error: operand 1 must be an integer or stack pointer register -- `add r7,r5@:r7=offset into ipackets for this input byte'
dvb/dvbsenco8.s:353: Error: operand 1 must be an integer or stack pointer register -- `and r7,r2@:mask r7 for wrapping'
dvb/dvbsenco8.s:354: Error: operand 1 must be an integer register -- `strb r8,[r1,r7]@:store the input byte into ipackets'
dvb/dvbsenco8.s:355: Error: operand 1 must be an integer register -- `adds r6,#-1'
dvb/dvbsenco8.s:358: Error: expected ZA array at operand 1 -- `ldr r1,=packetcount@:packet count'
dvb/dvbsenco8.s:359: Error: expected ZA array at operand 1 -- `ldr r2,[r1]@:get the packet count'
dvb/dvbsenco8.s:360: Error: operand 1 must be an integer or stack pointer register -- `add r2,#1@:count a packet processed'
dvb/dvbsenco8.s:361: Error: expected ZA array at operand 1 -- `str r2,[r1]@:save packet count'
dvb/dvbsenco8.s:363: Error: expected ZA array at operand 1 -- `ldr r1,=ipackets'
dvb/dvbsenco8.s:364: Error: operand 1 must be an integer or stack pointer register -- `add r0,r1,r5@:return with output packet address'
dvb/dvbsenco8.s:366: Error: unknown mnemonic `pop' -- `pop {r1,r2,r3,r4,r5,r6,r7,r8}'
dvb/dvbsenco8.s:367: Error: unknown mnemonic `bx' -- `bx lr'
dvb/dvbsenco8.s:369: Error: junk at end of line, first unrecognized character is `@'
dvb/dvbsenco8.s:387: Error: junk at end of line, first unrecognized character is `@'
dvb/dvbsenco8.s:1130: Error: junk at end of line, first unrecognized character is `@'
make: *** [Makefile:49: ../dvbrf] Error 1

In order to run properly, rpitx need to modify /boot/config.txt. Are you sure (y/n) Set GPU to 250Mhz in order to be stable
Installation completed !
```

zhudhjen, 2024-04-26:
I'm getting the same error message.

zardoz99, 2024-05-14:
Same here on RPI4 but using both "bullseye" and "bookworm" 64-bit OS releases.

zhudhjen, 2024-05-14:
I was using 64bit system as well, and I suspect this is the root cause. Will try 32bit system instead.

fhitchen, 2024-06-05:
I think it is due to the compiler version being incorrect on the RPI 64 bit OS.

I did this...
`$ sudo apt install build-essential gcc-arm-linux-gnueabihf`
and then in `/usr/bin`
`$ sudo ln -fs arm-linux-gnueabihf-gcc gcc`

And on to the next issue which looks like it is 64 bit related.

```bash
$ ./install
.
.
.
make: 'pift8' is up to date.
cc -Wall -g -O2 -Wno-unused-variable -c -o dvb/dvbsenco8.o dvb/dvbsenco8.s
cc -Wall -g -O2 -Wno-unused-variable -c -o dvb/dvbs2arm_1v30.o dvb/dvbs2arm_1v30.s
cc -Wall -g -O2 -Wno-unused-variable -c -o dvb/fec100.o dvb/fec100.c
g++ -std=c++11 -Wall -g -O2 -Wno-unused-variable -o ../dvbrf dvb/dvbrf.cpp dvb/dvbsenco8.o dvb/fec100.o dvb/dvbs2arm_1v30.o -L/opt/vc/lib -lrpitx -lm -lrt -lpthread
dvb/dvbrf.cpp: In function ‘int main(int, char**)’:
dvb/dvbrf.cpp:255:45: error: cast from ‘uchar*’ {aka ‘unsigned char*’} to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  255 |                 Result=_dvbs2arm_control (1,(uint32_t)IqBuffer) ;     // set DVB-S2 mode by passing buffer address
      |                                             ^~~~~~~~~~~~~~~~~~
make: *** [Makefile:52: ../dvbrf] Error 1
cc -Wall -g -O2 -Wno-unused-variable -c -o dvb/dvbsenco8.o dvb/dvbsenco8.s
cc -Wall -g -O2 -Wno-unused-variable -c -o dvb/dvbs2arm_1v30.o dvb/dvbs2arm_1v30.s
cc -Wall -g -O2 -Wno-unused-variable -c -o dvb/fec100.o dvb/fec100.c
g++ -std=c++11 -Wall -g -O2 -Wno-unused-variable -o ../dvbrf dvb/dvbrf.cpp dvb/dvbsenco8.o dvb/fec100.o dvb/dvbs2arm_1v30.o -L/opt/vc/lib -lrpitx -lm -lrt -lpthread
dvb/dvbrf.cpp: In function ‘int main(int, char**)’:
dvb/dvbrf.cpp:255:45: error: cast from ‘uchar*’ {aka ‘unsigned char*’} to ‘uint32_t’ {aka ‘unsigned int’} loses precision [-fpermissive]
  255 |                 Result=_dvbs2arm_control (1,(uint32_t)IqBuffer) ;     // set DVB-S2 mode by passing buffer address
      |                                             ^~~~~~~~~~~~~~~~~~
make: *** [Makefile:52: ../dvbrf] Error 1

fhitchen, 2024-06-22:
I think the problem is that the # character is the comment character for the ARM aarch66 assembler, but even if you change `@:` to `#:`  the assembly code is not compatible. I think it is X86 assembly code.
```bash
cc -Wall -g -O2 -Wno-unused-variable -c -o dvb/dvbsenco8.o dvb/dvbsenco8.s
dvb/dvbsenco8.s: Assembler messages:
dvb/dvbsenco8.s:67: Error: unknown mnemonic `push' -- `push {r14}'
dvb/dvbsenco8.s:71: Error: unknown mnemonic `pop' -- `pop {r14}'
dvb/dvbsenco8.s:72: Error: unknown mnemonic `bx' -- `bx lr'
dvb/dvbsenco8.s:99: Error: unknown mnemonic `push' -- `push {r1,r2,r14}'
dvb/dvbsenco8.s:101: Error: expected ZA array at operand 1 -- `ldr r1,=workbuff#:output to workbuff'
dvb/dvbsenco8.s:104: Error: expected ZA array at operand 1 -- `ldr r0,=workbuff'
dvb/dvbsenco8.s:107: Error: expected ZA array at operand 1 -- `ldr r0,=workbuff'
dvb/dvbsenco8.s:108: Error: unexpected characters following instruction at operand 1 -- `bl interleave#:returns with r0=pointer to an output packet
.
.
.
```

IdotMaster1, 2024-07-08:
Did some testing, you need to use a 32-bit version of Raspbian. Tested on a Raspberry Pi 4.

Janrupf, 2024-12-24:
If anyone else encounters this and doesn't want to ditch the 64bit OS for one reason or another, there is a ~very simple~ solution to get it working on a 64bit OS: use docker.

Dockerfile:
```docker
FROM debian:bookworm

RUN apt update && apt upgrade
RUN apt install -y curl gnupg
RUN curl -L http://archive.raspberrypi.org/debian/raspberrypi.gpg.key -o - | apt-key add -
RUN echo "deb http://archive.raspberrypi.com/debian/ bookworm main" > /etc/apt/sources.list.d/raspi.list
RUN apt update
RUN apt install -y git sudo build-essential
WORKDIR /src
RUN git clone https://github.com/F5OEO/rpitx.git
WORKDIR /src/rpitx
RUN ./install.sh
```

Build with: `docker build --platform linux/arm/v7 -t rpitx .` (note the `--platform`!)
Use it: `docker run --privileged -v .:/data --rm -it rpitx bash`