F5OEO/rpitx #326: POCSAG message sent 4 times and text is a alphanumeric string

INDIGI-CO-UK, 2024-04-28:
Hi,

I am getting 4 messages instead of one. I've tried to change the value to 1 in

https://github.com/F5OEO/rpitx/blob/master/src/pocsag/pocsag.cpp#L574 sets int REPEAT_COUNT = 4;

but no results - still getting 4 messages...

The second problem is that instead of the text I am getting some alphanumeric string. For example:
Text sent: 12345
Message received: 139- 86530

What is the reason for this?

VasylSamoilov, 2024-12-23:
De-duplication of messages is pager feature that needs to be enabled during programming. You can set repeat count via command line.  You need to also to match RIC code and slot message type on transmit. You are probably sending alphanumeric message to numeric ric or vice versa.

ZeGecko, 2026-11-04:
If it's a numeric only pager make sure to use numeric messages with "-n":
`printf '%s' "1234:12345" | sudo ./pocsag -f "433000000" -n -t 1`

If it's an alphanumeric pager, I had the same issue on one of mine. For some reason it was swapped around and function bits 3 and 2 were programmed for numeric messages, and function bits 0 and 1 were programmed for alphanumeric it seems. Try changing the function bit with "-b int":
`printf '%s' "1234:HELLO WORLD" | sudo ./pocsag -f "433000000" -b 0 -t 1`