F5OEO/rpitx #327: Continuous wave during NFM .wav file tx

martinpetrovaj, 2024-05-01:
While I was playing around with the NFM transmission test in easytest.sh, I noticed that rpitx will transmit a strong continuous tone on the center frequency for the entire duration of the transmission.

To test this, generate a .wav audio file with only silence (or use any signal / audio you prefer follow by / with intermittent silence) and use it with the provided NFM test.
When receiving on the same frequency your pi is txing on, you will notice a strong narrow signal that shouldn't be there.

This causes problems when transmitting digital data, or when a receiver (a 3rd party radio-controlled relay in my case) expects bursts to be separated by silence.

KubaPro010, 2024-05-01:
If i'm not wrong, you're talking about a carrier wave

martinpetrovaj, 2024-05-01:
That… might just be possible
_\*visible embarrassment\*_

Still, is it the norm to transmit the carrier wave even when the input audio / IQ is silent?
I suppose it's then up to the user (i.e. me) to turn off the transmission completely if I want to achieve a period of silence. Which is fine ig, though a bit impractical for short silent periods due to overhead in the code (esp. when using higher-level language wrappers), I'd just like to know if I'm correct or still missing something important.

Apologies if my questions are exceptionally dumb, trying to pick up and learn radio knowledge on the go

KubaPro010, 2024-05-01:
Yes, the carrier wave will stay if IQ is silent, take tune.cpp as a example, it justs enables the clock and sets the frequency (and power and advanced pll) but still, i don't see anything feeding any data in (unrelated: also a comment with the 2nd line, are we in character.ai or what?)

martinpetrovaj, 2024-05-06:
Thanks, went over the C++ code and more examples and I see it now.

It also seems that the signal I want to produce should be a bit burst ("time modulated" instead of FM) and so the original issue doesn't really make sense anymore. Thanks for steering me in the right direction a bit.

closed 2024-05-06 by martinpetrovaj