F5OEO/rpitx #328: Amplitude modulation

Michael-AUG, 2024-05-20:
Hi there

Firstly, what cool software!

Given you are able to produce both FM and SSB, is there any way that you'd be able to generate amplitude modulation with this software? When BBC R4 on longwave (198kHz) turns off, I am hoping I can use this, via some attenuation, to directly drive an old AM radio receiver.

Many thanks. 73 Michael GM5AUG

KubaPro010, 2024-05-21:
From what i can see, you in theory can generate AM, however the RPi cant control the output amplitude very well, and as AM has a lot with amplitude, that causes problems, SSB in theory should be the same but it somehow works (idk how, i haven't listened to rpi ssb, only fm)

Michael-AUG, 2024-05-21:
Interesting - thanks. I wonder whether @F5OEO would consider this as a future feature request?

KubaPro010, 2024-05-22:
No, you can't control the amplitude unless you do a hardware mod or the raspberry pi would get a better power output selection (right now, rpi 5 is not even supported in rpitx)

Michael-AUG, 2024-05-22:
OK. Thanks anyway - was worth asking :)