F5OEO/rpitx #330: Swedish letter Å Ä Ö

ericedberg, 2024-07-12:
Swedish letter Å Ä Ö does not work in Pocsag. Hopefully that could be fixed :)

mfalkvidd, 2024-07-12:
Pocsag uses 7-bit ascii. There is no ÅÄÖ in 7-bit ascii.

Did minicall have some tweak that added support for ÅÄÖ?

ericedberg, 2024-07-13:
If you scan and decode 161,4375mhz there will be message with ÅÄÖ.

mfalkvidd, 2024-07-13:
Cool. Should be fairly easy to reverse-engineer then. What software do you use for decoding? Multimon-ng?

ericedberg, 2024-07-13:
https://www.discriminator.nl/pdw/index-en.html

I use PDW and old school police-scanner modified  whit discraminator-tap connected to line-in/mic to a PC.

VasylSamoilov, 2024-12-23:
The way POCSAG works usually lowercase Engish letters are replaced with other alphabet. You need to build your own codepage. You also may need to use hex codes for some symbols. rpitx accepts and will send anything from 0x00 to 0X7F. Different pagers will have different codepage, there is no standard encoding in pocsag (except English), but when you switch language in pager firmware it no longer applies.

mfalkvidd, 2025-01-21:
Some info that might be useful in this issue: https://github.com/EliasOenal/multimon-ng/issues/69

VasylSamoilov, 2025-01-21:
This is correct. Character table (hex code received to character on screen mapping) is different for pager model+pager language combination. 
Example: motorola advisor+Swedish will be different from phillips messanger lux+Swedish, and some pager like motorola alpha bravo will only display English uppercase (cause it's screen is one-line limited character display).