F5OEO/rpitx #333: question about power output

DuckyPolice, 2024-08-06:
how many watts would this put out? i am trying to transmit on 20m and 40m and i have a good enough wire antenna, i was just wondering how far this would go in the average suburban area. i will be using pi zero 2 probably

dterweij, 2024-08-10:
not more than a few MiliWatts (0,00x Watts).

KubaPro010, 2024-08-10:
> not more than a few MiliWatts (0,00x Watts).

uh no, it is 52,8 milliwatt (3,3V * 0.016A)

penfold42, 2024-12-01:
> > not more than a few MiliWatts (0,00x Watts).
> 
> uh no, it is 52,8 milliwatt (3,3V * 0.016A)

uh no, not in a real world application it wont be.
You need to ignore/filter the harmonics and look at the power into the load.

434 MHz into 50 ohm I recently measured the channel power at 8.1dBm ~= 6.46 mW.

Using a 1/4 telescopic antenna with no ground plane I can easily transmit NFM audio 6 Km

imcalledfyre, 2026-10-12:
this is entirely dependant on your antenna. just using a pin gets you an ERP of a few nanowatts, theoretial maximum power output would be ~50 mW so you probably won't be able to even make a contact with this. Plus it has tons of spurs.