F5OEO/rpitx #335: Transmits, then stops

radio-satellites, 2024-09-15:
Hello,

I've been having an issue with rpitx. It transmits for a second, but then stops. The binary does not exit nor show any debug info on stdout. 

Any thoughts on why this might be happening? 

[image]

KubaPro010, 2024-09-18:
Have you set the `core_freq` in the `config.txt` file? (and also what about `force_turbo`?)

radio-satellites, 2024-09-18:
Yep, I followed the default installation script.

radio-satellites, 2025-09-07:
> Hi just following up, did #329 <https://github.com/F5OEO/rpitx/pull/329>
> solve this for you? It did not for me, and the config.txt was appended to

I believe so - everything seemed to start working normally again. I also
never had this problem again on any of my future installations.