F5OEO/rpitx #338: Unable to replay 27Mhz toy RC car

Hritik14, 2024-10-10:
Hello

Thanks for writing rpitx! It is really nice to be able to try out tiny experiments on a rasp pi.

### Setup 1: RTL SDR plugged into Raspberry Pi Zero 2 W for both capture and replay
I have been trying to replay a toy RC car signal but have been unable to do so. Initially, I tried the ``rtlmenu.sh`` script but there wasn't any feedback from the RC car while replaying the signal coming from its controller. 

### Setup 2: RTL SDR plugged into Mac and Raspberry Pi Zero 2 W for the replay
I captured the controller signal using my Mac and tried to replay the signal directly via ``sendiq``.

1. While replaying recorded signal:
[image]

2. The original signal that I recorded coming from the controller:
[image]

Here's a video showing the same: https://www.youtube.com/watch?v=iNaZDuqw-dI

I recorded the signal using this on my Mac (connected to RTL SDR):
```sh
; sudo rtl_sdr -s 2048000 -g 33 -f 25494052 myrec.iq
```

I replayed the signal using the following:
```sh
sudo ./sendiq -s 2048000 -f 25494052 -i ../myrec.iq -p 7 -l -t u8
```

``sendiq`` did mention about sample rate being too high, I tried with a couple of more sample rates but the least that RTL SDR supports is 250KHz and I was unable to get a good quality signal from the controller at 250KHz sample rate.
The docs do mention about having a band pass filter, but I suppose it wouldn't be a problem for an extremely short range transmission (not even 1 meter). I'm not sure what I'm missing.

Can someone help me with the direction where I should focus to debug this. I'm clueless as of now.

Hritik14, 2024-10-10:
I'm using dietpi as rasp pi OS. [This issue](https://github.com/F5OEO/rpitx/issues/337) suggests that gpu freq might also have to do something with it. I will try again with similar settings mentioned in the issue or a fresh installation of raspbian.

Ben-Does-Arcade, 2024-10-11:
Double check your `config.txt` to see if rpitx appended to the wrong file.

Should be at the bottom:

```
gpu_freq=250
force_turbo=1
```

Bandpass filters are generally a good idea but I've gotten away with not using them at all many times before. For low frequencies there might be a need for one. It doesn't hurt if you add one anyway just to filter out the extra noise.

What is the length of your antenna?

Hritik14, 2024-10-27:
I didn't have those lines in config.txt appended. I've freshly installed raspbian now. Those lines are appended correctly now.
Although, for some reason, no signal is coming out at all.
I tried:
1. Receive (view-only) the signal on my mac
2. Record the signal at the freq ``rtl_sdr -s 2048000 -g 33 -f 27160994 myrec.iq``
3. Send the iq file to rasp pi
4. Replay ``sudo ./sendiq -s 2048000 -f 27160994 -i ../myrec.iq -p 7 -l -t u8``
5. Nothing happens :(
6. Although the chirp test seems to be working properly.

Perhaps the recording was not good enough ? I'm not sure what has changed since the last time but something is wrong.
Here's a screen recording of the same: https://youtu.be/zqL8-tI9nNI

Hritik14, 2024-10-27:
Here's the recording I'm working with: https://drive.google.com/file/d/1hu1uy5KVONKozAuQBbFUfsuuKdYaPm6p/view?usp=sharing

Hritik14, 2024-10-27:
Here's a sample recording generated by 
```
rtl_sdr -s 2048000 -g 33 -f 27160994 myrec.iq
```
https://drive.google.com/file/d/1x2knrtrn3zeIfReZ_anUhY3_35Rto7jw/view?usp=sharing

Hritik14, 2024-10-27:
Something has to be wrong with the recording...
I tired the following:
1. Plug rtl sdr to Mac
2. Run ``AM easytest`` on rasp pi
3. Observe data being received on rtl sdr (mac)
4. Plug out rtl sdr from mac and plug it in to rasp pi
5. Record on rasp pi using ``rtl_sdr -s 250000 -g 32.8 -f 434M local_test.iq``
6. Play ``AM easytest`` on rasp pi again
7. Plug out rtl sdr from rasp pi and plug back to mac
8. Replay recorded ``AM easytest`` using ``sudo ./sendiq -s 250000 -f 434M  -i ../local_test.iq  -t u8``
9. *expect* data on mac, but actually nothing happnens

Again, here's the video: https://youtu.be/isfphOiQbEY

Either
1. ``rtl_sdr`` is not recording at all
2. or ``sendiq`` is not working

Hritik14, 2024-10-27:
ok... so something was wrong with ``sendiq``, i'm able to get somewhere using ``rpitx``.
I was able to reproduce the ``AM easytest`` by recording on my mac, transferring it to rasp pi and then replaying it from there.
Although... it is sped up for some reason... 
https://youtu.be/Tsd2e2-HUZE

Hritik14, 2024-10-27:
I was able to slow down the audio by decreasing sample rate from 250k to 100k at the replay time, it made the audio "sound" slower but had lost most meaning... not gonna work this way for the RC car